CurtRod/SimpleEVSE-WiFi #14: Button yes/no?

njordan77, 2018-07-23:
One short question before i add this element. What if i do not consider a button? will it start when plugged in or it waits until the GUI activates the controller?

What if i use the button and activate and somebody else presses the button again....will it disable again? if this is the case its a little dangerous if i leave the car in public and e.g. kids playing next to this nice box and turn it off "accidentially"....

Thanks a lot for clarification, Norbert

CurtRod, 2018-07-23:
Hi Norbert,

Thank you for your Message! SimpleEVSE-WiFi waits to activate until the button is pushed, reading a RFID tag or activating via WebUI.
Deactivating works the same. So yes, it could be dangerous, when kids playing next to the Box! ;-)
A possible solution could be a flag in the Settings like "allow deactivating via Button". Maybe this will be delivered with the next release.

You can follow the discussions about SimpleEVSE-WiFi also on goingelectric.de (german):
https://www.goingelectric.de/forum/ladeequipment/simpleevse-wb-mit-wlan-t31260.html

Curt

closed 2018-07-23 by njordan77

njordan77, 2018-07-25:
One question regarding the hardware part of this button.
Should the button bring GND to the configured GPIO pin? should this be done simply via connecting to GND....or should this be done via PULL-UP Resistor in combination with the button in order to have stable 3,3V when not pressed at the pin.

How about debounce, is there something implemented in Software or its expected to happen in Hardware? Regards Norbert

CurtRod, 2018-07-26:
Hi Norbert,
as described in README.MD the button should bring the GPIO to GND. The ESP8266 Module has internal Pull-Up resistors, which are used here. There is no hardware Pull-Up necessary.
Debounce is already prevented in Software.

Curt

Macrisu, 2019-07-23:
The original design asks for an auto-reset button.
I used that button with ESVE-Wifi and it works only if the connection is closed (I keep pressing the button).
It can be made to work with that button, or do I need to buy a classic on/off button?