CurtRod/SimpleEVSE-WiFi #15: No authentication possible on Safari MacOS and iOS

michel72, 2018-07-28:
It's not possible to authenticate on iOS and MacOS because apple devices do not allow basic auth without https.

In short, it's not possible to use the webUI from and Apple IOS device at all and It's not possible to use the webUI on a Apple Macbook using the Safari webbrowser. It does work on a Macbook when using Chrome or Firefox.

njordan77, 2018-08-01:
can confirm this....on Win10 everything connects perfect to APmode....but iphone does show the login mask but from there it does not continue with correct password into the main GUI.....is there a way to fix this?

njordan77, 2018-08-14:
Any progress here, or solution that could fix the issue?

CurtRod, 2018-08-14:
Sorry for late response. I will try to find a solution asap.

Leftaf, 2018-08-28:
I found following https://github.com/Links2004/arduinoWebSockets

and Issue https://github.com/Links2004/arduinoWebSockets/issues/25

may this help to improve security to iOS

njordan77, 2018-11-11:
Hi, any update? Tried recently the latest version + hoped naively that IOS12 could have fixed it - but no success. Still not able to connect in AP Mode from my Iphone. Connecting with my PC (Win+Firefox) is no problem at all. Regards Norbert

CurtRod, 2018-12-25:
I suspect the error in the JS part of the web interface. Unfortunately, I do not have a Mac to hit the debugger in Safari. So at the moment I can not analyze what exactly the problem is. Does anyone have the opportunity to do this? That would be great.

njordan77, 2018-12-25:
hi, i would love to support - but the only thing i have in hand is an iphone 6s - if that may help. Thanks for the support!

michel72, 2018-12-26:
[image]

michel72, 2018-12-26:
Safari does not allow basic auth for websockets.

https://serverfault.com/questions/910702/apache2-safari-exclude-basic-authentication-for-websockets

michel72, 2018-12-27:
@CurtRod , if you need any help testing, I have an iMac, macbook, iPhone and iPad available.

CurtRod, 2018-12-27:
@michel72 , thank you. I wondered, because i've seen some other projects using websockets with basic auth and there are no problems with safari. I have only an iPhone 7 for testing here but there is no way to debug with that.
**good news for all Apple users:**
In the next release will be an option to deactivate the basic auth, however this should only be a work around and not a preferred solution.