CurtRod/SimpleEVSE-WiFi #17: Contribution: STL files for a simple 3D-printed DIN-rail enclosure for the WeMOS D1 mini

FrenkK, 2018-08-08:
I made a simple DIN-rail enclosure design for the WeMOS D1 mini.
It's made of two halves that are screwed together using recessed-head wood screws. There is (just) enough space to install a simple 5V power supply cannibalised from a phone charger. The antenna is exposed for best possible WiFi reception.
Build:
1. For safety, print using fire retardant plastic.
2. Connect all parts and use a drop of superglue to hold them in place in the enclosure.
3. Screw the enclosure together using 4x 3,6x16 mm wood screws with a recessed head.

See the picture for the final look (rightmost module).
[image]
[STLs.zip](https://github.com/CurtRod/SimpleEVSE-WiFi/files/2269021/STLs.zip)

Macrisu, 2019-08-15:
Thank you for the 3D STL files.
I see that you also put a LED and button on the DIN case, very nice.
You use the button to activate the charge?

I've tried this, but it does not seem to work for me. I have an auto-reset button that works only if I keep the button pressed manually.
**Your button is on/off or auto-reset?**

Thank you

FrenkK, 2019-08-16:
> You use the button to activate the charge?
No, this is a fallback in case the SimpleEVSE-WiFi fails. It is a momentary NO switch corrected directly to the control input on the Simple EVSE Wallbox. If there is ever a big problem with the Simple EVSE Wifi, I can just turn off its power and use the wallbox on it's own.

Macrisu, 2019-08-16:
Do you have the original cad files for STL?
I want to make the WeMOS D1 mini support smaller.

FrenkK, 2019-08-17:
Sure, I added them to the original post.