CurtRod/SimpleEVSE-WiFi #18: switch to STA mode is failing only emergency AP Mode get's activate.

Leftaf, 2018-08-13:
It looks like that there is an issue to switch from AP mode to STA after once AP has been Setup successful.

Trying to go into STA mode always the WEMOS start in fallbacktoAPMode();
[image]
as this function is only called in the Setup(); sequence there must be an issue with loading the config.json
[image]

Does any body else have this behavior?

in depend of this behavior the [debug] activation is unfortunately after the relevant error messages :-( so the code should be move upwards and the first messages should be always shown.
`````
bool ICACHE_FLASH_ATTR loadConfiguration() {
  File configFile = SPIFFS.open("/config.json", "r");
  if (!configFile) {
    Serial.println(F("[ WARN ] Failed to open config file"));
    return false;
  }
  size_t size = configFile.size();
  std::unique_ptr<char[]> buf(new char[size]);
  configFile.readBytes(buf.get(), size);
  DynamicJsonBuffer jsonBuffer14;
  JsonObject& json = jsonBuffer14.parseObject(buf.get());

  if (!json.success()) {
    Serial.println(F("[ WARN ] Failed to parse config file"));
    return false;
  }
  if (json.containsKey("debug")){
    debug = json["debug"];
  }
  if(debug){
    Serial.println("Debug Mode: ON!");
    Serial.println(F("[ INFO ] Config file found"));
    json.prettyPrintTo(Serial);
    Serial.println();
  }
  else{
    Serial.println("Debug Mode: OFF!");
  }
```

CurtRod, 2018-08-14:
Thank you for reporting this issue. I will have a look at this in the next days!

CurtRod, 2018-08-28:
okay. I changed the lines a bit diffent, but now there would be an error message on Serial, when no config file is found. Thank you for your help.

closed 2018-08-28 by CurtRod