CurtRod/SimpleEVSE-WiFi #20: Reconnect or connect if use timer

autodiag, 2018-11-18:
I have a bit of a problem with using in a while when I have a timer setting (to run at a low tariff) so the charging is not started, is the problem of the system itself or is the problem elsewhere?
if it's a system problem, would it be good to run timers and run automatically at the set time?

the kind of problem is that if I press the button on the car connector, I will not be charged as I have any other chargers and I have to re-enter the phone (or the button)
I have a 680R resistor (for 20A limit) PP to GND ..
Is this involved well or do I have to engage PPs on the car?
it was not quite clear to me from the description

Martin

autodiag, 2018-11-18:
i use J1772 and car is Fiat 500e

CurtRod, 2018-12-25:
Hey Martin, 

sorry for the late reply, I was a little longer on vacation. :)
The 680R resistor in the cable should not be a problem. 
Do I understand correctly that you are using the timer function of your vehicle? There will be a timer function directly in SimpleEVSE-WiFi with one of the next releases. Maybe that would be the easier way. Is SimpleEVSE-WiFi in the status 'activated' when the vehicle requests the charging depending on the vehicle's pre-setting?

Curt

autodiag, 2019-01-17:
when I have a timer set in my car due to a cheap night's charge, the EVSE does not charge the car, it will not start ... just start if the timer is off, I load the charger and charge ... but if I pull out the charger and reinsertion or if the electrified site will drop out of the car and it will not restart again .. it will be possible to run the manual manually via the web interface ... it might not be wrong to choose whether to plug in the car or to start up manually, that would solve the problem, I think along with the timer to help the charger, not the car

Matin