CurtRod/SimpleEVSE-WiFi #24: Feature Request: stop charging when power/current below defined value (stop charging at 80%)

badsoden, 2019-03-04:
It would be nice if one could set a treshhold where the charging will be halted when power/current gets below defined value. The idea is to set that value where the SoC is about 80%. Not all cars support charging stop at a certain SoC value and this way the EVSE could manage this. It increases lifetime of the battery and allow full recuperation right after start driving. E.g. Renault ZOE does not have an option to charge at 80% and the battery must be well below ca. 90% to have full recuperation. So charging at 80% is good for battery and lowers your average consumption. 
For each car the owner shuld first estimate/assess the actual current at 80% SoC and then set the treshold accordingly.

FrenkK, 2019-03-04:
AFAIK, the car does not send a feedback about state of charge to the EVSE with SAE J1772 signaling protocol, so this is not really feasible without some additional hardware, at least not accurately.
The recuperation is a valid argument, but that aside, I question if it makes sense to do this. Car batteries are used in a much narrower voltage window than consumer batteries you can find e.g. in phones. In my Opel Ampera, for example, only 10 out of 16 kWh capacity is used and cell voltage never goes above 4 V (as opposed to 4,2 V in consumer batteries). Decreasing the charge window still further will probably not make much difference and you should already get at least some 5000 cycles of life. It may be a factor with some cars, depending on their use of the battery, although for sure not a big one.

badsoden, 2019-03-04:
I though so, that there is no feedback about SoC. That's why I suggested to use the charging curve. The car reduces charging current at around 70 to 80% with most cares and I though I use that to estimate SoC. I thought in its most simple form one would just be able to set minimum current and when it goes below charging stops. It most have a short waiting time before it kicks in to ensure the car takes maximum power (above the treshhold). E.g. logic would be: after start charging wait one minute and then if current below xyz then stop. 
That 80% is better for the battery I derived from Tesla that normally stops at 80% unless you explicitly set the flag to charge 100%. That some manufacturers put more capacity in the car than actually seen is true. In my ZOE I though it is another 2 or 4 KWH on top of the specified 22 KW. 

Recuperation is especially valid if one lives uphill. If you charge till 100% and drive down after charging then that last 5% to 10% of charge is basically wasted because one can't recuperate.

FrenkK, 2019-03-04:
Yes, the charging curve approach should work, if you have an energy meter connected. It is not trivial, though, because it must take into account the set current that can change with dynamic load shedding, for instance.

badsoden, 2019-03-04:
That is very true, one probably has to average out the (s0) measured values over some minutes and then shut off charging. If one loads 79 or 81% is not very critical anyway so a few minutes earler or later is not important. Some basic rolling robin algorithm should do the job.

Macrisu, 2019-07-20:
for LEaf, It can be done with an Android APP.
https://play.google.com/store/apps/details?id=com.chk.lcc&hl=en

premultiply, 2019-11-22:
The way to estimate the SOC from the charging power does not work reliable.
Charging Power measured on EVSE side depends on many factors like battery temperature, pilot signal, battery voltage, preconditioning state, battery heating, ...

It would end in may problems and failures.

Apart from that it is kind of useless as the BMS keeps track of the battery health itself and 100% displayed SOC for user does not mean the battery cells are fully charged on most types of cars, especially ZOE.

badsoden, 2019-11-22:
I know the ZOE doesn't report real SoC. My main issue is, that recuperation is more or less disabled when you start driving with a 100% battery. In fact I don't mind when the battery is loaded somewhere between 75 and 95 %. I just want to sty away from the 100%. I also not so convinced that Renault's BMS is very accurate with loading. Renault is quite buggy with all of their SW hence why would their BMS be an exemption. So, if unreliable means one day (e.g. when cold) it would stop at 75% and another day in summer it would charge to 90% I can live with that, recuperation will work right from start and the battery will definitely not suffer from that. Actual current measurements can come from S0 sensor or RS485 bus from separate power meter. 
Is it worth it to make this effort on the EVSE for a ZOE only? Not sure obviously and not up to me to judge or answer. ZOE is a popular car but if it is the only EV with such limited charging control then I understand it's low on the priority list. I really don't know how many cars do not allow to set the end of charging at a certain level of SoC. The facelifted ZOE (2019) doesn't allow it either. Their App to track charging or SoC is not working any more which I found all quite disappointing. One could overcome it by setting a timer but this is quite cumbersome as one can set a clock but not a timer.  The environmental gain of a EV does depend on the final life span of the battery and ZOE is not really making an effort here it seems.

premultiply, 2019-11-23:
Why dont you have a look for OpenWB? All these kind of advanced features are included there.

But this project here should hopefully stay smart, small and clean.