CurtRod/SimpleEVSE-WiFi #25: MQTT for EVSE-WIFI

roesl, 2019-04-12:
Is it possible to add MQTT feature to Control the EVSA wit MQTT commands ?

Georg

s0170071, 2019-09-16:
+1 from me. 
I do already have the mains power readings and solar power readings in my OpenHab installation.   
I would like either
a) communicate these power readings to EVSE so it can use it for solar charging
b) control power output of EVSE myself (based on my readings)
MQTT style is very much appreachiated.

neographikal, 2020-02-16:
+1

This would make it much more versatile!

alx2k, 2020-02-25:
@neographikal, It would be much more versatile, but it is not needed.

I have "home assistant" at home, and a wemosEM power meter (also on github). When my home consumption changes in 1 Amp, home assistant sends a http request to the charger using "http://192.168.4.1/setCurrent?current=XXXX". It works flawlessly.

@s0170071, If you want, I can send you my home assistant configuration, maybe it is similar in Openhab.

neographikal, 2020-02-25:
I can get it to work, but MQTT is much more versatile, reliable and almost everything I use is connected through MQTT. Especially when it's a ESP32/8266 (Tasmota/esphomelib), for some stuff I even wrapped the existing API for MQTT with Nodered.

Thanks for the suggestion but this still would be important to me :)

spoeh-man, 2020-05-11:
+1 for me is also better with mqtt for iobroker

njordan77, 2020-05-13:
i would also love to see your great project compatible with MQTT (pubsubclient, rather easy to handle)

kolaCZek, 2020-06-24:
+1 Do want!!!

krispek, 2020-07-07:
+1 for me. The HA-solution works, but MQTT would make it even better. :)

captain186, 2020-10-15:
> If you want, I can send you my home assistant configuration, maybe it is similar in Openhab.

Hi, I'm intrested in using evse-with home assistant. Could you send me your configuration, please?
Thanks a lot in advance!

kolaCZek, 2020-12-01:
```
rest_command:
  evse_enable:
    url: 'http://192.168.0.140/setCurrent?current=13'
  evse_disable:
    url: 'http://192.168.0.140/setCurrent?current=0'

switch:
  - platform: template
    switches:
      wifi_evse:
        value_template: "{{ is_state('sensor.wifi_evse', '13') }}"
        turn_on:
          service: rest_command.evse_enable
        turn_off:
          service: rest_command.evse_disable
```

alx2k, 2020-12-01:
Sorry for the delay, I did not see Github notifications. My code changes the current depending on the main house energy meter consumption (working with a WemosEM).

It reads the house energy consumption on the meter, and sustracts that consumption from my contract maximum amperage (34 Amps), so I don't get any energy cuts due to amperage overlimit.

Code is:
`rest_command:
    simpleevsewifi: 
        url: http://192.168.1.40/setCurrent?current={{ states('sensor.Amperios_Cargador') }}

sensor:
  - platform: mqtt
    state_topic: "wemos/wemosEM-4568CC/power"
    name: Amperios Cargador
    icon: mdi:power-plug
    unit_of_measurement: "A"
    value_template: "{% if 34 - value_json.current | round(0) > 32 %}
    {{32 | round ()}}
    {% else %}
    {{34 - value_json.current | round(0)}}
    {% endif %}"
`

captain186, 2020-12-07:
Thanks a lot! I'll try it out!