CurtRod/SimpleEVSE-WiFi #27: Metering just doesn't work

fsg4u, 2019-05-11:
Hey,
Great project and the charging worked nearly out of the box but it just can't neater the Power with the S0 counter.
It stays at 0.00kW the whole time.
I checked the S0 pins and it drags 3.3V down to 0V every 0.5Wh and the pulse is supposed to be 80ms. 
Any Idea what could be wrong?
Thx

twityde, 2019-05-24:
Did you check the polarity?
It is important how you connect the S0.
S0+ to GND, S0- to D3/GPIO1.

fsg4u, 2019-05-24:
Hey, Yes I did. Didn‘ work either but I just changed the GPIO and now it’s working. Perhaps it’s just that the GPIO is broken

hgmendes, 2019-08-26:
I'm having this problem now.
S0+ connected to D3 and S0- to GND, but stays at 0.00 kWh.

Meter doesn't inform impulse lenght, but I've tried with 80 and 60ms.

Anyone has this working with an Orno OR-WE-507 meter?

CurtRod, 2019-08-30:
Did you try to change the polarity?
Many other Orno electricity meters have a pulse length of 90ms. For the OR-WE-507 I could find no exact information. If necessary, ask the manufacturer directly.

hgmendes, 2019-09-02:
Hi,

Thank you for your reply.
I've already asked manufacturer but until now theres no answer :/

Regards,

Hugo Mendes
GSM PT: +351 916 903 578
Skype: hgmendes

hgmendes, 2019-09-10:
Solved! Works ok with 90ms pulse! Thank you!

closed 2019-11-22 by CurtRod