CurtRod/SimpleEVSE-WiFi #32: Charge won't start

hgmendes, 2019-07-16:
When i press 'Activate EVSE' it won't start charging, 'Deactivate EVSE' turns green after some seconds. I could hear contactor closing and opening in a few seconds interval. Reseting all parameters to default (with no SimpleEVSE-Wifi) charger is working fine.

Macrisu, 2019-07-16:
Plz give details on full setup.
Some pictures with the setup can help.

Are some registers modified from default?

hgmendes, 2019-07-16:
Hi, thank you for your reply.

I don't have pics right now, but I could send tomorrow.

I have already tried in several SimpleEVSE WB and also in a SimpleEVSE Din.
Setup is only SimpleEVSE (firmware 11) and Webos D1 Mini. D1 and D2 to
TX/RX, 5V and ground to VCC and ground on SimpleEVSE. All parameters in
default.

Regards,

Hugo Mendes
M: +351 916 903 578

Macrisu, 2019-07-17:
My EVSE Din is still on the post so I'll try your pain soon.
I'm working with NodeMcu, as I hate to solder pins. 

=====
NOTE#1: By default, MODBUS is NOT enabled, so the original analogue input switches can
be used (current limit and current boost functions). MODBUS can be enabled by pulling
AN input down to GND while booting (= when you power the device up) at least 5 times
within 3 seconds (button activated). This will change register 2001 to the value 1,
but it will not save the value permanently. _The value will be saved after the first successful R/W
operation over MODBUS (register number >=2000)_.
=====

**Are you sure that Modbus is enabled permanently?**

1. Verify if your PINS are soldered correctly on Webos D1 Mini.
2. 
D1-TX
D2-RX

This can be mixed by mistake!

2. The documentation for EVSE is a mess, as many have changed over different version of FW + DIN version has si own pinouts.
Also, the new HW has Mobus active by default, but you should try to verify the default registers, as I suspect that there is the main problem.

For my setup (kW, WIFI and RFID and Type 2 soket), I plan to check the Registers:
2000 - 32
2001 - 1
2003 - 0
2004 - 0
2005 - b0-0, b1-0, (I do not plan to use the button)
2007 - 0 (I'll use a socket, so I want to read the resistor from the cable).

https://www.evalbo.de/download-1/

Some useful stuff

Macrisu, 2019-07-18:
On the german forum, somebody had the same issues and the problems seemed to be D1 (defective)
https://www.goingelectric.de/forum/viewtopic.php?t=31260

Macrisu, 2019-07-18:
Well...
I try to activate Modbus - 5x press on the button on the first 3 seconds after power-on, but no success :(

I get Modbus error in the WIFI interface.

hgmendes, 2019-07-18:
Normally Modbus error is D1/D2 > TX/RX crossed or Modbus register not
activated. Try to do it faster.

Hugo Mendes
GSM PT: +351 916 903 578
Skype: hgmendes

Macrisu, 2019-07-18:
I'm getting mad with this...
I cannot activate modbus :(

Also, in the EVSE WIFI Settings, I get no info about Device Status
Also, the LOG is always empty
[image]

hgmendes, 2019-07-18:
What is the version of SimpleEVSE firmware? Not all versions have modbus
activation on boot. Does the LED stays on for a few seconds on boot or
start immediately to blink?

Device status only appears when modbus connection is ok.

Hugo Mendes
M: +351 916 903 578

Macrisu, 2019-07-18:
It works...
I've forgot to connect the GND wire :(

All working great now, even RFID 💃

hgmendes, 2019-07-26:
How are you powering the NodeMCU? I think my problem could be there, since
i'm powering it from SimpleEVSE 5v...

Hugo Mendes
GSM PT: +351 916 903 578
Skype: hgmendes

Macrisu, 2019-07-26:
I’m using an US chromecast adapter 5v, 1a
I finally got an use for US and UK phone adapters...

Bogdan Gheorghe
+40743188002

Macrisu, 2019-08-17:
Well, I have your problem now, and the culprit is the power supply.
It says 1A 5V and worked for few days, but now I have disconects after few minutes.
I’ll use a Well Din Rail power supply via usb.

hgmendes, 2019-08-26:
Hi,

My problem was solved connecting 5v from ac/DC converter to esp, I presume that 5v delivered by SimpleEVSE wasn't enough.

closed 2019-11-22 by CurtRod