CurtRod/SimpleEVSE-WiFi #33: How to reset WIFI settings?

Macrisu, 2019-07-20:
Hello!
If for some reason I need to reset the ESVE-WIFI settings to default (Access Point), how can I do it?
Is there a hardware way?

I need to move my box go another location and I need to change the WIFi SSID to connect.
If I forget to reset the AP-Client mode, how can I access the WIFi interface on the new location?

CurtRod, 2019-08-30:
If the configured WiFi is not found in client mode, EVSE-WiFi starts in fallback mode. In fallback mode, EVSE-WiFi starts as AP and restarts every 10 minutes. Within these 10 minutes, you can use the web interface to configure your new WiFi in client mode.

Macrisu, 2019-08-31:
Now I know that via trial and error.
Thank you for reply.
Also see please the problem with the button.

CurtRod, 2019-09-04:
Do you mean the button issue, which is discussed in goingelectric.de?

Macrisu, 2019-09-04:
Yes, that issue..

Bogdan Gheorghe
+40743188002

CurtRod, 2019-09-04:
The Button must be connected as described in README.md:
NO -> D4 (GPIO2 on ESP)
COM -> GND
You have to activate the button also in the settings page in the WebGUI of EVSE-WiFi.

CurtRod, 2019-09-08:
Did it work?

closed 2019-11-22 by CurtRod