CurtRod/SimpleEVSE-WiFi #34: Port forwarding problems?

Macrisu, 2019-09-10:
I try to access my EVSE from the internet. I've forwarded the external port 1234 to port 80 of my EVSE.
I can access the login page, but nothing is happening when I press the login button.

Do I need to do something else? To forward another port?

Macrisu, 2019-09-16:
I've forwarded only the TCP port.
Do I need to forward the UDP port, also?

hgmendes, 2019-11-04:
Hi Macrisu,

Forwarding TCP 80 and UDP 80 port i can access from outside...

However would be interesting that we could define in which port SimpleEVSE Wifi is listening for that we can use a different port rather than 80 since this port could be in use for other appliances also or even for the router, and some routers couldn't be configured with other port rather than 80.

rtfmjoey, 2020-01-11:
I'm using a reverse proxy and in addition to standard http protocol it's also needed to proxy the websockets protocol. The UDP protocol isn't needed.

Example:

<IfModule mod_ssl.c>
	<VirtualHost *:443>
		ProxyRequests Off
		ProxyPreserveHost On
		ProxyVia On

		ProxyPass /ws ws://192.168.40.31/ws
		ProxyPassReverse /ws ws://192.168.40.31/ws

		ProxyPass / http://192.168.40.31/
		ProxyPassReverse / http://192.168.40.31/

		ServerName evse.domain.example
		ServerAdmin webmaster@domain.example

		ErrorLog "/var/log/apache2/evse.domain.example-error_log"
		CustomLog "/var/log/apache2/evse.domain.example-access_log" common

		SSLCertificateFile /etc/letsencrypt/live/evse.domain.example/fullchain.pem
		SSLCertificateKeyFile /etc/letsencrypt/live/evse.domain.example/privkey.pem
		Include /etc/letsencrypt/options-ssl-apache.conf
	</VirtualHost>
</IfModule>

Using a Let's Encrypt SSL certificate the EVSE is now accessible via https from the outside world.

mkrasselt1, 2020-09-28:
Its not the port thats Causing the Problems, Press F12 in your Browser and Check the Konsole for Warinings and Errors. The should be some "insecure websocket not allowed outside personal networks" of some sort which is cuased by the browser rejecting the (because the lack of power in the esp8266) insecure websocket connection over the internet. of course you can disable the feature but @rtmfjoey 's solution is way better

Macrisu, 2022-03-27:
> I'm using a reverse proxy and in addition to standard http protocol it's also needed to proxy the websockets protocol. The UDP protocol isn't needed and the password is not sent unencrypted over the net.
> 
> Example:
> 
> ```
> <IfModule mod_ssl.c>
> 	<VirtualHost *:443>
> 		ProxyRequests Off
> 		ProxyPreserveHost On
> 		ProxyVia On
> 
> 		ProxyPass /ws ws://192.168.40.31/ws
> 		ProxyPassReverse /ws ws://192.168.40.31/ws
> 
> 		ProxyPass / http://192.168.40.31/
> 		ProxyPassReverse / http://192.168.40.31/
> 
> 		ServerName evse.domain.example
> 		ServerAdmin webmaster@domain.example
> 
> 		ErrorLog "/var/log/apache2/evse.domain.example-error_log"
> 		CustomLog "/var/log/apache2/evse.domain.example-access_log" common
> 
> 		SSLCertificateFile /etc/letsencrypt/live/evse.domain.example/fullchain.pem
> 		SSLCertificateKeyFile /etc/letsencrypt/live/evse.domain.example/privkey.pem
> 		Include /etc/letsencrypt/options-ssl-apache.conf
> 	</VirtualHost>
> </IfModule>
> ```
> 
> Using a Let's Encrypt SSL certificate the EVSE is now accessible via HTTPS from the outside world. Just forward port 443 to the reverse proxy and you're good to go.

I've done this and I can access the EVSE via HTTPS
The problem is now, that I can not log in!
After entering the password, the WIFI connection is lost for some reasons ;(

[image]

No such problem via LAN

schenlap, 2022-07-04:
I discovered a problem regarding port forwarding. The evse website always connects to websocket port 80/ 443, regardless of website port (may be different for forwarding).

```
function wsConnect() {
  var protocol = "ws://";
  if (window.location.protocol === "https:") {
    protocol = "wss://";
  }

  wsUri = protocol + window.location.hostname + "/ws";
  websock = new WebSocket(wsUri);

```

Maybe https://www.w3schools.com/jsref/prop_loc_port.asp could help to get port of current url and use this for websocket connection (to also use the forwarding).

mkrasselt1, 2022-07-09:
i think this is because insecure websocket is not allowed on https websites - do you use reverse proxy for adding https ?

Macrisu, 2022-07-09:
I tried with reverse proxy https and without reverse proxy, simple port fwd.

Same problem.

mkrasselt1, 2022-07-16:
could you use a vpn?
As i wont be that confidend to expose this web-api to the public