CurtRod/SimpleEVSE-WiFi #35: Charge stops when ventilation start

hgmendes, 2019-09-10:
Hi,

I've implemented this version in an ESP8266 and when charging Renault Zoe's the charge stops when AC starts to cool the batteries. I've already tried with almost 10 Zoe's and the problem is always the same. If we use SimpleEVSE as provided without modifications problem doesn't occur, problem only occur when power off by software is activated. As when Zoe start to cool batteries the charge is paused for a few secondsand then won't reactivate! 
I'm thinking to develop a kind of timer for that charge can restart... any comments?

hgmendes, 2019-10-02:
Any comments regarding similar cenario? Any one else with Zoes?

CurtRod, 2019-11-04:
Sorry for the late reply. Is it correct, that Zoe stops charging for a few minutes when ventilation starts? There is a timer checking, if the vehicle is already charging. It would be possible, to set up this timer, so the charging will be interrupted by EVSE-WiFi later. Can you tell me how long Zoe interrupts charging when ventilation starts?

hgmendes, 2019-11-04:
Hi Curt,

Thank you for your reply.
The Zoe could take from 10 to 60 seconds to reactivate charging after
ventilation start.
I have feedback now that also occurs with other models like Kia e-Niro and
Hyundai Kauai.

Hugo Mendes
GSM PT: +351 916 903 578
Skype: hgmendes

CurtRod, 2019-12-15:
can you reproduce this error with Version 0.3.0? Is there a difference in "normal mode" or if you using "always active"?

hgmendes, 2019-12-17:
Hi Curt,

I've didn't yet made the update, i've just made a change to 2005 register
to 0 and problem has stopped, but i've lost the RFID activation function...

I will try to update this weekend and i will let you know.

Regards,

Hugo Mendes
GSM PT: +351 916 903 578
Skype: hgmendes

hgmendes, 2019-12-19:
Problem continues even on the new 0.3.0 version.
eGolf also have problems to charge with RFID activation, when 2005 register set to 0 it possible to charge without problem

CurtRod, 2019-12-20:
did you also try the "always active mode"?

hgmendes, 2019-12-21:
Yes, in “always active mode” works fine. Problem only occurs when using
cards or button (in app or physical) to activate EVSE.

premultiply, 2020-08-17:
RFID authorisation and other internal parameters should only be reset on car connect/disconnect but never when just contactor switching is done with cable connected.

DavidMenting, 2021-01-11:
I can confirm that the Hyundai Ioniq is affected by this issue as well. Charging stops after a few minutes in normal mode. In always active mode charging works fine altough it is then impossible to start or stop the charging session.