CurtRod/SimpleEVSE-WiFi #38: ESP32 support

NOWherers, 2019-11-27:
Hi, thanks for your so far!
I just wanted to kindly ask about the ESP32 support.

Is the integration of the ESP32 still ongoing?
Can someone help with it?

best regards, bas

CurtRod, 2019-11-29:
Integration of ESP32 will be completed in early 2020. I guess it will be january or february. But I have to create a new project, because the libraries for the ESP32 are too different. Do you want to help me with that?

MarcoV-git, 2020-03-01:
Hey Curt,

Can you please share an update about the status of ESP32 support?

Marco

CurtRod, 2020-03-11:
Hey @MarcoV-git 

ESP32 support is still in progress an should come with 0.5.0 in a few weeks.

Curt

CurtRod, 2020-07-26:
now, ESP32 took more time than expected, but now ESP32 support is released

closed 2020-07-26 by CurtRod

HA-TB303, 2020-12-30:
@CurtRod Hi,
I respect your efforts for implementing the esp32 support and to offer that on your own hardware, but as a hardware tinkerer I would rather build myself. Is there a public version for esp32 planned, along with documentation on which GPIO's are used?

Macrisu, 2020-12-30:
Same request here:
Gpio map and firmware upload.
I think you can even make a software pack with the info to buy for who don’t want to use your hardware.

HA-TB303, 2021-01-11:
HI @CurtRod , any comments on this?