CurtRod/SimpleEVSE-WiFi #40: Charge duration not correct when using 'always active'-mode

rtfmjoey, 2019-12-15:
When using the 'always active' mode of version 0.3.0 the charge duration is not calculated correctly because 'millisStopCharging = millis();' is only executed when _not_ in always active mode.

After changing line 782-785 from 

`        if (vehicleCharging) {  // EV interrupted charging
          vehicleCharging = false;
          toDeactivateEVSE = true;
        }
`

to 

`
        if (vehicleCharging) {  // EV interrupted charging
          vehicleCharging = false;
          toDeactivateEVSE = true;
          millisStopCharging = millis();
        }
`

calculation is done correctly in always active mode.

CurtRod, 2019-12-15:
I've seen that too. Will be corrected in the next few days with update to 0.3.1.

rtfmjoey, 2020-01-11:
Issue was resolved succesfully in 0.3.1

closed 2020-01-11 by rtfmjoey