CurtRod/SimpleEVSE-WiFi #41: [Question] How to update via webui?

HA-TB303, 2019-12-15:
Sorry, no issue here, just a question:

Can someone confirm that this webui update procedure is correct?

1. Download the lastest release zip file: https://github.com/CurtRod/SimpleEVSE-WiFi/releases
2. Extract the zip-file
3. Go to http://[ipaddress], select settings and upload firmware
4. Select file (from unpacked zip file): /bin/latest/built.bin 

Reboot.

rtfmjoey, 2019-12-15:
Yes, that's the correct procedure, did it a few times like this.

Please note that when clicking 'Upload' a spinning wheel with 'Uploading' will appear and just a blank page with 'OK' when uploading is finished. The esp will reboot automatically.

CurtRod, 2019-12-15:
Yes, that's correct!

closed 2019-12-19 by CurtRod