CurtRod/SimpleEVSE-WiFi #42: Enhancement: Disable/Enable EVSE when using always on-mode

rtfmjoey, 2019-12-15:
Is it possible to add a feature to allow enable/disable of EVSE via API when using always-on mode?

I've integrated with Domoticz to balance charging current and when max current is exceeded the EVSE is disabled so I don't blow the main fuses. 

In always-on mode this is not possible atm, the EVSE is not deactivated when setStatus?active=false is used.

CurtRod, 2019-12-15:
That is correct. In always active mode EVSE will never be deactivated. This feature may come with version 0.4.0.

CurtRod, 2020-03-15:
use the API to set Current to 0:
`http://YOURIP/setCurrent?current=0`

closed 2020-03-15 by CurtRod

andig, 2020-08-13:
@CurtRod coming from https://github.com/andig/evcc I'm confused which API to use. In the past I've used `setStatus` which still seems to exist but you're recommending `setCurrent`. When would I use the one or other?