CurtRod/SimpleEVSE-WiFi #43: Evse-wifi 0.4.0 said password?

Macrisu, 2020-03-14:
I’ve upgraded to 0.4.0 version yesterday (now is a newer version).
After update, my wifi is now password protected on the ssid!

Is very hard for me to access the module.
What is the wifi pass? It was initially open (no wifi pass)

Macrisu, 2020-03-14:
Now I know how to erase a NodeMCU.

You need to install Python on Windows 10.

https://www.digitalocean.com/community/tutorials/how-to-install-python-3-and-set-up-a-local-programming-environment-on-windows-10

After installation, run those commands in Powershell Admin mode:

```
python -m pip install --upgrade pip
pip install pyserial
pip install esptool
```

The command to erase the flash is:

`esptool.py -p com5 erase_flash`

Modify com number as needed in your case.

This will reset the ESP8266 to defaults.

CurtRod, 2020-03-15:
I don't have any idea why there was a password set. I tried to reproduce this issue but could understand this bug.

bauer1980, 2020-03-15:
Yes, It askes for username and password after logging in with adminamin - after the first startup. I can't change and access settings.

Macrisu, 2020-03-15:
Use Chrome, as the safari option from settings doest not seems to work.

bauer1980, 2020-03-15:
great,
thank you for the quick support.

yes, it didn work with safari and firefox, but with chrome.

bauer1980, 2020-03-15:
hi, sorry to disturb again,
it only works at the first login,
unfortunately i interrupted,
and now it also asks me with the chrome for a user and a password after using „adminadmin“ at the login page.

thank you for any suggestion!

best regards,
Wolfgang

Macrisu, 2020-03-15:
Use refresh in Chrome.

It will go away.

In the settings deactivate the basic auth.

CurtRod, 2020-03-18:
did this solve the problem?

bauer1980, 2020-03-18:
i am not sure. i used chrome, on the phone and the laptop and tried it many times and suddenly it worked. After deactivate the basic auth. for safari use, it works without problems.

mpolak77, 2020-04-26:
I think my issue is similar - I‘ve updated to 0.4.x and suddenly I cannot associate to the
evsewifi AP anymore. It used to be wpa2 before but now it seems my original WiFi password does not work or something else is broken. Since the upgrade it seems to be running in always on mode