CurtRod/SimpleEVSE-WiFi #44: change evse register with web ui - change modbus value

bauer1980, 2020-03-16:
Hello,
i can access and change evse wifi settings, i can also see evse register on "show device status" - but i have the problem that i cannot change the register values. If i set them they don't change. Also after restart they are not changed. 
I have on register 2001 following value: 25973, i want to change it to 1 but it does not work. I always get modbus error on the main page. I checked all wire connection, i cannot find the problem.
Also if i put the AN pin 5 times to 0 within the first 3 seconds of start i don't get anything changed.
[image]

bauer1980, 2020-03-17:
Issue solved: I didn't connect GND --> GND from EVSE DIN to the ESP8266

CurtRod, 2020-03-18:
sorry, for the late reply. It looks like there is no connection to EVSE DIN. It also should show you the "Modbus error" on the "EVSE Control" Page.

closed 2020-03-18 by CurtRod