CurtRod/SimpleEVSE-WiFi #46: JSON api confusion

andig, 2020-04-11:
I'd like to support SimpleEVSE-WiFi as part of https://github.com/andig/evcc.

Reading https://www.evse-wifi.de/dokumentation/#HTTP_API I find the JSON structure not matching the code (https://github.com/CurtRod/SimpleEVSE-WiFi/blob/master/src/main.cpp#L1002). According to the code, all elements are prefixed with `evse_` and not using camel case.

Since I don't have a SimpleEVSE Wifi for testing, could you please clarify which JSON structure can be expected?

mkrasselt1, 2020-09-28:
i have version 0.4.1 and it outputs:
`{"type":"parameters","list":[{"vehicleState":1,"evseState":true,"maxCurrent":20,"actualCurrent":10,"actualPower":0,"duration":0,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0,"mileage":0,"meterReading":25.18,"currentP1":0,"currentP2":0,"currentP3":0}]}`
i dont expect this to be changed in later versions

andig, 2020-09-29:
Perfekt, thank you!

closed 2020-09-29 by andig