CurtRod/SimpleEVSE-WiFi #50: Limited to 3kW?

Macrisu, 2020-05-27:
Hello!
I have a leaf 2019 40 kWh, which is using only one phase and max 6.6 kW to charging => 32A tree phase.

This is my second EVSE, but the first with a cable.
I've put a 220 OHM resistance between PP and PE for 32A on the cable

[180924 analoger Steuereingang Simple EVSE WB.pdf](https://github.com/CurtRod/SimpleEVSE-WiFi/files/4691771/180924.analoger.Steuereingang.Simple.EVSE.WB.pdf)

I use the Modbus version.

The problem is that I'm limited to 3 kWh, as reported by the car.
I've tried with another EV station and I get 6 kW.

Where should I start looking?

premultiply, 2020-05-27:
There have to be matching PP resistors on both ends of the cable. One on car side and one on EVSE side. There is no wire connection between them!
Both, car and EVSE detect the cable capabilities on "their" resistor. Thar will not drew more current than allowed and the EVSE will not signal a higher current as allowed by its resistor as PWM to the car. So there is double safety.

On other EVSE-systems the resistor on EVSE-side may be replaced by internal register setting only but the resistor on car's side is always requiered.

Macrisu, 2020-05-27:
As I said, what both ends of the cable?
I have one 220 OHM resistor inside the male connector that enters the car.
The other side of the cable is connected to EVSE.

See the PDF on the first post.

hgmendes, 2020-05-27:
You also need to put a 220ohm resistor between PP connection and ground on
SimpleEVSE.

Hugo Mendes

Macrisu, 2020-05-27:
Even if I use modbus?
I need to put that 220 ohm resistor on evse?

[image]

premultiply, 2020-05-27:
Yes. As written on both ends if the cable.
The car needs one and the EVSE needs one.

Macrisu, 2020-05-27:
I’ve put the second resistor on evse gnd-pp and is working now!
Than you all
[image]

closed 2020-05-27 by Macrisu