CurtRod/SimpleEVSE-WiFi #53: API calls from different devices cause non-responsiveness

FrenkK, 2020-06-11:
I use Home Assistant to check the EVSE status and a separate program to set charging current according to available power.
The Home Assistant readout works well on it's own and charging current setting via API works as well, when they are working separately. But if both are working, the system stops responding to everything, both webpage server and API responses until I reset.
I assume it must be a problem when API call happen at the same time and this causes something to break.

mkrasselt1, 2020-09-28:
the small webserver of the esp cant handle more then one concurrent session - consider lowing the polling intervall and try routing the output of your program back through home assitant - mine is using an automation to set the power evertime the inverter gives me new power readings

FrenkK, 2020-09-29:
This is what I did, but it seems that the problem should be handled better.
I understand the hardware limitation, but the error should be handled more gracefully instead of having to reset everything each time.

mkrasselt1, 2020-09-30:
you actualy dont have to reset it - because when the resources are free again, it starts responing again - if not you are flooding this smal thing with requests

FrenkK, 2020-09-30:
It does recover on it's own, but it takes a long time.