CurtRod/SimpleEVSE-WiFi #61: Ladung startet manchmal nicht mit 2.0.4

schenlap, 2020-11-22:
Seit ich auf 2.0.4 upgedated habe startet die Ladung nicht. Normalerweise verwende ich evse im Remote mode mit evcc. Wenn es sort hängt bringt auch ein umschalten auf den Normalmode nichts.

[image]

Auch nach mehreren Minuten startet die Ladung nicht. Erst nach mehreren Reboots (zwischen Remote- und Normalmode umschalten) hat die Ladung gestartet.

schenlap, 2020-11-22:
`{"type":"parameters","list":[{"vehicleState":2,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":0,"duration":3066,"alwaysActive":true,"lastActionUser":"API","lastActionUID":"API","energy":0,"mileage":0,"meterReading":0,"currentP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}`

schenlap, 2020-12-07:
Heute ist es das erste mal auch im Echt-Betrieb aufgetreten. Nur die Box neustarten hat das Problem behoben. Wenn man über die API einen Stromwert vorgegeben hat habe ich eine korrekte Antwort erhalten, es ist aber nichts weiter passiert.
`S0_set current to given value`

CurtRod, 2020-12-29:
sorry für die späte Rückmeldung. Besteht das Problem nach wie vor? Welchen Stromzähler verwendest du denn?

schenlap, 2020-12-30:
Es ist jetzt schon einige Zeit nicht mehr aufgetreten, ich werde weiter beoabachten.
Stromzähler ist intern ein S0 Zähler.

schenlap, 2021-03-23:
Ist jetzt schon lange nicht mehr aufgetreten

closed 2021-03-23 by schenlap