CurtRod/SimpleEVSE-WiFi #62: API - getLog Crash

JFK344, 2020-12-26:
Get Log is not working when loging is disabled or the log is empty. The Problem ist that the ESP will crash and not recover.

CurtRod, 2020-12-29:
thank you for reportig this bug. Will be fixed in one of the next updates

JFK344, 2021-01-02:
Thank you! I have developed a IoBroker Adapter for the ESVE Wifi.  [https://github.com/JFK344/ioBroker.esvewifi]

edsub, 2023-01-31:
Thus issue is still present I think?

Macrisu, 2023-01-31:
Last update for this free version was a long time ago so…
I’ve used Etek chinese controller and accesories and is very stable.
If you need PV control, I recommend Go-e charge with phase change.

JFK344, 2023-01-31:
> Last update for this free version was a long time ago so… I’ve used Etek chinese controller and accesories and is very stable. If you need PV control, I recommend Go-e charge with phase change.

? who said anything about pv controll ?