CurtRod/SimpleEVSE-WiFi #66: Recompile 1.x version to work on ESP32 ?

Macrisu, 2021-02-19:
Hello,

It is possible to modify/recompile version 1.x to run on an ESP32?

https://maker.pro/esp8266/tutorial/how-to-upgrade-from-esp8266-to-esp32

CurtRod, 2021-02-28:
You can compile the existing code also for ESP32.

Macrisu, 2021-02-28:
I will try...
From what I see here, Arduino is not needed?
https://github.com/CurtRod/SimpleEVSE-WiFi/pull/65

HA-TB303, 2021-03-03:
That sounds great, I need the same for my little DIY charger DIN Rail thingy ;)
[image]

Macrisu, 2021-03-04:
@hoermto I used NoddeMCU and for some reason, they are crap and cannot connect to them after firmware upgrade.
Also, WIFI is very bad.
I've ordered two ESP32-WROOM-32U and WIFI antenna.

If you can compile the 1.x firmware for ESP32 it would be great.

hoermto, 2021-03-04:
I am using a NodeMCU 8266 currently. Wifi is bad, yes. Other is that it has no free Pins to extend with Oled or CPInt.
It will need some time on my side to get it done, however, I have the same plan. Using a ESP32 with external antenna to use in my SimpleEVSE v1.

hoermto, 2021-03-04:
> 
> 
> That sounds great, I need the same for my little DIY charger DIN Rail thingy ;)
> [image]

For this case, what is you requirement? You can compile the code for the Wemos D1 mini still ... if you migrate to ESP32, you pin layout will change, so I suspect you cannot use this hardware directly.
In reverse, you wont be able to get all EVSE-WIFI v2 features into the 8266 because of limited pin count and 8266 memory shortage (e.g. OLED libraries won't fit into the firmware). There are ways, like if you dont have a RFID, you could use these pins for a CPInt (requires code changes). 
There is a high number of combinations of ESP versions across different features which might be connected in different ways (e.g. EVSE connected via software-serial or HW serial). The maintainer of this project has a clear scope of the software, which might stay in conflict with other peoples ideas.

So, regarding your HW setup, what would be your requirement? Just being interested :-)

HA-TB303, 2021-03-05:
Hi, when using the wemos d1 mini (esp8266) the following GPIO's are exposed on the terminals:

D1 true D8.

That translates to the MH-ET Live esp32 as:
Wemos D1 mini -> MH-ET-live esp32
D1 (GPIO20) -> GPIO22
D2 (GPIO19) -> GPIO21
D3 (GPIO18) -> GPIO17
D4 (GPIO17) -> GPIO16 
D5 (GPIO14)-> GPIO18
D6 (GPIO12)-> GPIO19
D7 (GPIO13)-> GPIO23
D8 (GPIO15)-> GPIO5

[image]

HA-TB303, 2021-03-06:
Hi @hoermto we found a better PSU for the job and are in the process of redesigning the PCB.  It's a universal Wemos D1 mini / MH-ET live esp32 design, but If you can let me know what GPIO's would be ideal for the esp32 version of the Simple-EVSE-Wifi, we will expose those!

This design will be shared on our website in due time, so everybody can build it themselves.

Concept design:
[image]

HA-TB303, 2021-03-06:
SPI is definitely not doable within this form factor. The display would have to be I2C driven (SDA/SCL). If the code could be adjusted to that, I could have GPIO 26, 18, 19, 23, 5, 17 and 16 free for other stuf, along with TX and RX.

hoermto, 2021-03-06:
You D1-D8 exposed. It depends on what meter you use and what meter you want to use, also whether RFID is used or not.
Minimum is D1/D2 for EVSE connection when not using Modbus. With the rest you might use OLED, CPInt and button (or LED) if you dont use RFID. If you use a Modbus Meter, you could use the LED and Button, since the D3 for S0 is released. If you are using a Modbus over RS485, you also could connect the EVSE over Modbus, and then have another D1/D2 release.

There are many options. It all depends finally from the hardware layout. If you want to remain pin-compatible between D1-mini and D1-mini-32, only D0-D8 + TX/RX from D1-mini and their equivalents of D1-mini-32 are available.
As an additional option, the TX and RX pins should be possible to to used aditional digital in/out as well. With this you can add more features to your layout. However, this would require to run EVSE via D1/D2 (not Modbus), and would require commenting out the modbus meter from the code (currently always enabled, even if not used).

Here is the default PIN layout for the 2 versions available (Simple-EVSE 1.0 and 2.0 / SmartWB). If you can compile yourself, feel free to comment out what you don't need and reassign pins. As long as we cannot freely define the pin assignment via config, the compiled firmware follows this layout. Especially the change of the EVSE from Modbus to Serial and reverse is tricky if you are not familiar with all this.

For myself I am thinking to add some more config options to the software. However, compatibility to SimpleEVSE needs to be made sure in case it should have a chance for a merge.

Actually, as far as I know the EVSE-DIN module at least (the others as well I think) expose a 5v pin you could use for the EVSE-WIFI module. Use the free space for a relais for the CPInt probably? [http://evracing.cz/evse/evse-wallbox/evse-wb-din_20190712.pdf](url). 

# pins EVSE-WIFI

|Module|Module PIN|SimpleEVSE 1.0<br/>Wemos D1 Mini|SimpleEVSE 2.0<br/>Wemos D1 Mini32|GND|3.3v|Remark
|:----|:-:|:-----:|:-----:|:-:|:-:|:------------------------|
|LED EVSE-WIFI|+|D0|26|x||optional
|EVSE|TX|D1|- (*4)|||using SoftwareSerial
|EVSE|RX|D2|- (*4)|||using SoftwareSerial
|EVSE|A|||||using RS485 bus A
|EVSE|B|||||using RS485 bus B
|SDM120<br/>SDM630|A|||||using RS485 bus A
|SDM120<br/>SDM630|B|||||using RS485 bus B
|RS485|TX|TX (*1)|TX|||Modbus interface
|RS485|RX|RX (*1)|RX|||Modbus Interface
|S0|data|D3||x||||So meter
|Button (Charge)||D4|16|x||optional
|Button (Reset)||Rst|Rst|x||optional
|RFID|Base|||x|x|
|RFID|SCK|D5|18|||(*2)
|RFID|MISO|D6|19|||(*2)
|RFID|MOSI|D7|23|||(*2)
|RFID|SDA|D8|5|||(*2)
|RFID|RST||22|||(*3) optional
|CPInt|S||4|x|x|
|RSE|||2|||(*5)
|Oled|cs||12||||
|Oled|dc||13||||
|Oled|reset||33||||

(*1) in SimpleEVSE 1.0 only for Modbus meter required. Currently occupied in any case.   
(*2) RFID uses always first ictl pins from board. Cannot be changed  
(*3) no pin on Wemos D1-Mini/8266  
(*4) in SimpleEVSE 2.0 and SmartWB the EVSE is connected via Modbus only  
(*5) "Rundsteuerempfänger" (receiver for external control), SmartWB only. Extra hardware required

HA-TB303, 2021-03-06:
Hi, thanks for elaborating this. At the chance of sounding really stupid now, in which file are the pins defined?

hoermto, 2021-03-07:
I did analyze the code and made this table for me in order to be ready for changes, which don't break the original design. 
Maybe I can inject this into the repository later so it will be default documentation.

HA-TB303, 2021-03-07:
Ok thx. I’ll beter wait untill the open source support for the esp32 is more documented and supported then.

cola32, 2021-04-17:
@HA-TB303 where can I get such a board ?

HA-TB303, 2021-04-17:
Hi, We will publish a video and article on https://espthings.io soon where you can then get them made on pcbway.

cola32, 2021-05-14:
Cool thanks!

is it possible to use a version on an esp32 d1 mini?

Langer75, 2021-07-18:
Hi

one remarks: 
- one pin is missing, what is the RST of the RFID? (like for oled pin 33)
Did you elaborate this too?