CurtRod/SimpleEVSE-WiFi #67: meterReading liefert doppelten Wert

schenlap, 2021-03-23:
Hi,

wenn die Ladung beendet wird erhöht sich der meterReading Wert um die geladene Energie, siehe https://github.com/andig/evcc/issues/788
[evcc_20210323_meterReading.log](https://github.com/CurtRod/SimpleEVSE-WiFi/files/6191900/evcc_20210323_meterReading.log)

Eingesetzte Version: 2.1.1

schenlap, 2021-03-23:
energy stimmt, ebenso die Anzeige in der GUI.
Ich verwende einen S0 Zähler

schenlap, 2021-04-09:
aus dem evse-Beta Forum

`Edit: Der von dir genannte Bug wird tatsächlich mit der v2.2.0 behoben.`

closed 2021-04-09 by schenlap

thdietrich, 2023-07-21:
Von dem hier geschilderten Problem bin ich auch betroffen (Eingesetzte Version 1.1.1, letzte verfügbare Version für den ESP8266).

Gibt es eine Chance, dass eine neue Firmware für den ESP8266 mit dem Bugfix veröffentlicht wird? 
Ich konnte das Problem und die Lösung im Quellcode (Zeile 2710ff in Version 2.2.2 in main.cpp) nachvollziehen
mir fehlt aber aktuell die Entwicklungsumgebung, um die Firmware zu erzeugen.

pjpeartree, 2024-04-15:
Hi,

Same, I am also facing the same problem on  version 1.1.1, last available version for the ESP8266, as reported at [https://github.com/evcc-io/evcc/discussions/13299#discussioncomment-9121240](url)

Any change to get the fix for ESP8266?

Regards,
Pedro

schenlap, 2024-04-16:
@pjpeartree 
There was no activity in this repo for a log time.  Maybe you should report the bug on https://board.evse-wifi.de/viewforum.php?f=8