CurtRod/SimpleEVSE-WiFi #71: Interrupt CP Signal automaticly or by API

mkrasselt1, 2021-08-09:
We use remote controlled charging through homeassitant and excess solar engery,
but sometimes the car wont wake up automaticly so a CP interupt pressing is needed.
But manualy doing this removes all the usefullness of the api - so it would make a hell lot of sense
if the evse interrupts the cp signal eigther: permanently until charging starts, automaticly when charging starts after cars is present for x time or make it available through the api if possible.
i would add this myself but i dont have much time at the moment :/

Nebukadneza, 2021-11-26:
Hi,

check here:
https://github.com/CurtRod/SimpleEVSE-WiFi/blob/master/src/main.cpp#L3067
you simply go `curl -Xget http://<your-evse>/interruptCp` :tada:

mkrasselt1, 2021-11-27:
Thanks a lot, i updated and its working now :)

closed 2021-11-27 by mkrasselt1