CurtRod/SimpleEVSE-WiFi #73: Improve WIFI Signal - SOLVED

Macrisu, 2021-09-01:
My biggest problem with the original WEMOS D1 Mini or Node MCU was the poor WIFi signal.

Now I've found and tested this, with external antenna.

WEMOS D1 Mini Pro 4M 16M Bytes External Antenna Connector NodeMCU Based ESP8266 ESP-8266EX CP2104 WIFI Development Board Micro

https://www.aliexpress.com/item/32801063577.html
It's much better and it's working!

[image]

kirk-loeten, 2021-09-02:
I use the same Wemos in other projects too. It is necessary to solder the bright from onboard antenna to the connector for the extern antenna.

Macrisu, 2021-09-02:
Usually yes, but i this case no need to solder.

https://www.aliexpress.com/item/32801063577.html
[image]

kirk-loeten, 2021-09-02:
https://hackaday.io/project/165061-solar-powered-wifi-weather-station-v20/log/162011-adding-an-external-antenna

Macrisu, 2021-09-02:
Uau, thank you! I did not know about this!
It seems really stupid to sell an external antenna and never say that in order to use it, you must modify the board.

edsub, 2024-01-31:
@Macrisu Did you succesfully flash the V1.1.1 firmware to a Wemos D1 mini pro (with 16M)?
If so:
1. Do you still encounter frequent crashes/reboots of the ESP when running? Especially when accessing the log?
2. What esptool parameters did you use to flash?
I flashed the v1.1.1 with that standard flash.bat to a Wemos D1 mini pro (with 16M) but still encunter frequent crashes/reboots. I exoect the reasoning is that with flashing a 4M bin to a 16M board effectively made my board act as a 4M . . . .

edsub, 2024-01-31:
[image]

Macrisu, 2024-01-31:
I don’t use this anymore.
Too unstable and now I can buy a Made in Germany charger, 11kw and 6.5m cable for 200 EUR.
It simply does not make sense to use Simple EVSE.
I have 3 boxes and some controllers.