CurtRod/SimpleEVSE-WiFi #76: unstable Wemos D1 mini build?

Langer75, 2021-12-26:
I downloaded the latest binary for Wemos D1 mini and flashed it. Unfotunatelly it is unstable. It crashes and restarts again. Unfortunatly it is not a systematic crash. It is random. It happens more often in acces point mode, but also sometimes in client mode. Does anybody knows this  issue? Does I made something wrong?

Macrisu, 2021-12-26:
Bad hardware. Try another Wemos.

Langer75, 2021-12-26:
that was my first guess too. But I have two Minis and both show the same behaviour. I bought a third one and also this is unstable. When I flash tasmota or espurna the Minis are stable. It is just with the binary V1.1.1. May I try an older Version.

hgmendes, 2021-12-26:
What’s your power source? Recommendable value is 1A

Langer75, 2021-12-26:
V1.1.0 is also not stable. My power source is 5V 2A. When I measure voltage and current it is 5.16V and less than 100mA. Does it makes a difference whether a EVSE is connected or not? For test puposes the evse is currently not connected. So I got Modbus error. But this should not crash the Wemos

Langer75, 2021-12-26:
[image]
These are my D1 Minis. One is 12F, one is 8266mod 12F and one is 8266mod only

Langer75, 2021-12-26:
By the way. Also with connected EVSE it crashes

hgmendes, 2021-12-26:
Maybe something is not ok with your power source, the two versions are very
stable..

Langer75, 2021-12-27:
I checked several 5V  power adapters. With all the same behavior. I have no further ideas. Which flasher tool do you use?

hgmendes, 2021-12-27:
Hi use the ESPtool in the project to upload bin but there are a lot of
similar softwares.

Langer75, 2021-12-28:
To exclude my flashing hardware as failure I let my friend flash one of my D1 mini. And after he has flashed the D1mini it was also unstable with evse-wifi firmeware. A lot of other binaries are working (e.g. tasmota, ESPeasy, ESPurna) without resets. But all of them have smaller binaries. Could it be, that the ROM has a defect, that only has an effect with the evse-wifi binary? It would be a bad fluke, that I have three Wemos D1 minis, bought seperatly with several month/years in between and all of them have the same error. Has someone made similar experiences. Is it somehow possible to check the ROM?

Btw.: I have ordered another one

royspeed, 2022-01-22:
I too have had problems with a Wemos D1 mini. Without being connected to the EVSE DIN everything seemed fine, but after connecting pin D1 D2 and GND, the green led on the EVSE is dim. With also 220V voltage on the EVSE the Wemos froze. 
With different 5V 3A power supply the problem seems to be solved.

Macrisu, 2022-03-14:
I had a similar issue today when I've upgraded from NodeMCU to D1 Mini Pro with an external WIFI problem.
For some reason the D1 mini Pro clone does not power from the 5v pin and I had to use a cut cheap micro-USB cable from my 2.4A Power Source.

The cheap micro-USB cable was little long and was **close** to the 40A relay. This that poor shielding and I've got random crashes. I've moved the cable a few centimetres away from the 40A relay and no more problems.

edsub, 2024-04-23:
Did any of you solve this? i still have random crashes. I've read that the cause of this might be a too large firmware for the ESP. But since I am using an 8mb version and the memory usage in the gui not showing a teal shortage I think that should not be it.
I am using a micro-usb adapter to power the ESP.
[image]

rtfmjoey, 2024-04-23:
Are you using DHCP or static IP? 

I've experienced frequent crashes until I switched from DHCP to static IP.

Langer75, 2024-04-23:
indeed I use static IP

edsub, 2024-12-16:
I am on static IP as well.

tux21, 2025-07-06:
Hi gyus, does anyone found the successful solution of this issue? I bought an evse module, several wemos D1's and oled, but I was not able to find a way where to connect the oled and it crashes after few seconds no matter what.. Project seems to be abandoned by the author so what was your solution? Dump it all and start with something else? I was pretty happy how it works and HA implementation, but it is very unreliable. It even stops charging once it crashes.

royspeed, 2025-07-07:
I could never get it stable either, and now, years later, it's still sitting in a box waiting. In the end, I decided to go with a SmartEVSE instead: https://github.com/dingo35/SmartEVSE-3.5.